foostan/crkbd #1: corne-chocolate

AlexanderElias, 2018-09-15:
Awesome project exactly the design I have been looking for. Where might I find the design files for the corne-chocolate? Is it the classic?

jeffreykxiao, 2018-09-16:
I don't think corne-chocolate has been created yet. About a month ago, @foostan added boilerplate for crkbd++ and crkbd+h (a next gen ckrbd and a hotswappable crkbd), but more recently, he renamed the original crkbd to ckrbd-classic and set up a new folder for crkbd-cherry. I think we'll see designs for crkbd-chocolate in the next month.

If you're interested in the crkbd that supports kailh choc, then the classic is your best bet at this moment. The classic doesn't have or support hot-swap, though.

AlexanderElias, 2018-09-16:
Thanks @jeffreykxiao I appreciate the response. Looking forward to the chocolate @foostan thanks for the work. I think I will give the classic a try for now.

closed 2018-09-16 by AlexanderElias

foostan, 2018-09-18:
@jeffreykxiao Thank you for following me!
Corne-chocolate is still in development, I will release as soon as possible.

AlexanderElias, 2018-09-18:
Well count me in on this follow fiesta!