foostan/crkbd #2: Documentation not readily accessible

cschmautz, 2018-09-16:
## Issue  
Documentation for assembly is only in Japanese, which is fine if using Google Chrome which can do translations to English; however, the documentation is still very sparse, leaving first-time-user of both the LED display and SMD LEDs kind of in the dark about debugging the assembly of the board.  
  
### Symptoms  
My board, assembled, works just fine for key presses, but the LED screen and LEDs do not illuminate. Nowhere in the documentation really lists how to go about fixing / debugging these issues, just 'solder it in and it'll be good to go'. This is my first rodeo with these components and I'm a bit at a loss quite frankly.  
  
### Proposed solution  
Please give some English documentation, as well as possibly expand on troubleshooting steps to follow when these devices fail to respond. This can be housed in this repository, or if there is some other 'source of truth', please link to those documents for us newbs of the truly custom keyboard.  
  
Looking forward to hearing back with more information - have enjoyed the build so far!

jeffreykxiao, 2018-09-16:
I am assuming you're coming from /u/network_operations / keyhive's group buy? I feel like the support for the GB should really be from him, and the translation as well. Aside from that, you'll definitely get more support from posting to /r/olkb or /r/mechanicalkeyboards, especially if you include pics of your build and a link to the keymap you're using. I know it's linked in the crkbd documentation, but have you taken a look at the [Helix](https://github.com/MakotoKurauchi/helix/tree/master/Doc) docs? They're not the same board, but Helix has a lot of the same information. Maybe it can answer some of your questions.

Regarding the OLED screens - after looking at the default keymap from CRKBD, as long as you flashed it without any changes, it really should just 'be good to go'. There's a define in the config.h and some other stuff going on in the keymap so it shouldn't be an issue of `OLED_ENABLE = false` or anything. The only thing I would make sure is that you soldered the OLED jumpers on both halves of the keyboard (but not both sides of each pcb!)

As for the RGB LEDs, those SK6812minis are super finicky, as in they have a high failure rate if exposed to heat over 240(?)C and as a side note they don't respond well to isopropyl alcohol or deionized water. I think those are enabled in the default keymap, but I'm not super sure. If the first (closest to the pro micro) LED is dead, then then none of the other LEDs would work. Best to use a multimeter to check em out. I also dug up an old tweet from mtei / MakotoKurauchi, creator of the helix, regarding testing of those LEDs, hope it can be useful to you: https://twitter.com/mtei/status/970334819400888321

foostan, 2018-09-18:
Thank you to your feedbacks.

> Documentation for assembly is only in Japanese

I will plan to translation in english, but I should development new pcb and more.
Please wait the moment.

heinrisch, 2018-12-08:
> Thank you to your feedbacks.
> 
> > Documentation for assembly is only in Japanese
> 
> I will plan to translation in english, but I should development new pcb and more.
> Please wait the moment.

How is the translation going?

foostan, 2018-12-11:
@ryo1kato Thank you so much, You've been very helpful.

closed 2018-12-11 by foostan