foostan/crkbd #4: OLED bitmap creation

nicolai86, 2018-11-10:
Hey @foostan ,

Thank you so much for this awesome keyboard - I've build it yesterday and it works very well.
I was wondering if there is any documentation on how to create a custom logo for the OLED? 

I've found an example on how to do it for the ErgoDox Inifinity OLED but the approach leaves the crkbd oled blank :(

Thank you so much for this keyboard.

Raphael

jeffreykxiao, 2018-11-10:
@nicolai86 I was under the impression that since this uses a really similar OLED setup to the Helix's (ssd1306.h/c present in both repos), the instructions for Helix should apply here. 

Give [the Helix firmware guide](https://github.com/MakotoKurauchi/helix/blob/master/Doc/firmware_en.md#edit-font-data) a shot, if you haven't already.

foostan, 2018-11-13:
> but it didn't work unfortunately

Really?
Please show me error messages and logs.

nicolai86, 2018-11-14:
@foostan So I'm trying to use the helix [FontConverter](https://github.com/MakotoKurauchi/helix/tree/master/FontConverter)

When I run it on a 128x32 bmp file I get 

```
python img2hex.py /tmp/aws.bmp
0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
0x00, 0x00, 0x00, 0x00, 0x00, 0x70,
0x78, 0x3C, 0x3C, 0x3C, 0x1E, 0x1E,
0x1E, 0x1E, 0x1E, 0x1E, 0x1E, 0x1E,
0x3E, 0x7E, 0xFC, 0xFC, 0xF8, 0xF0,
0xE0, 0x00, 0x00, 0x00, 0x00, 0x1C,
0xFC, 0xFC, 0xFC, 0xFC, 0xF0, 0x00,
0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
0x00, 0x80, 0xF8, 0xFC, 0xFC, 0xFC,
0xFC, 0xFC, 0xF0, 0x00, 0x00, 0x00,
0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
0xF0, 0xFC, 0xFC, 0xFC, 0xFC, 0x1C,
0x00, 0x00, 0x00, 0xC0, 0xF0, 0xF8,
0xF8, 0xFC, 0x3C, 0x3E, 0x1E, 0x1E,
0x1E, 0x1E, 0x1E, 0x1E, 0x1E, 0x1E,
0x3C, 0x3C, 0x38, 0x00, 0x00, 0x00,
0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
Traceback (most recent call last):
  File "img2hex.py", line 31, in <module>
    b = 1 if im[fontw*rowc+fontwc][fonth*(colmax-1)-fonth*colc+fonthc][0]<128 else 0
IndexError: index 128 is out of bounds for axis 0 with size 128
```

When I run it on a 64x32 bmp I get a similar error: 

```
0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
0x18, 0x18, 0x0C, 0x0C, 0x0C, 0x0C,
0x1C, 0xF8, 0xF8, 0xE0, 0x00, 0x00,
0x78, 0xF8, 0xE0, 0x00, 0x00, 0x00,
0xC0, 0xF8, 0x38, 0xF8, 0xE0, 0x00,
0x00, 0x00, 0x80, 0xF8, 0xF8, 0x18,
0x00, 0x40, 0xF0, 0xF8, 0x9C, 0x0C,
0x0C, 0x0C, 0x0C, 0x18, 0x00, 0x00,
0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
Traceback (most recent call last):
  File "img2hex.py", line 31, in <module>
    b = 1 if im[fontw*rowc+fontwc][fonth*(colmax-1)-fonth*colc+fonthc][0]<128 else 0
IndexError: index 64 is out of bounds for axis 0 with size 64
```

I've tried pasting the above matrix anyways into `logo_reader.c` but the display just remains black if I do that.

nicolai86, 2018-11-18:
I tried using the image from logo_reader.c and generate an image with this: http://javl.github.io/image2cpp/ , but I can't get it to generate the image displayed on the OLED... :(

nicolai86, 2018-11-21:
Okay, slowly making progress. I can feed the font from `glcdfont.c` into http://javl.github.io/image2cpp/ , select `128x80` and `read vertical` 
[image]

This means I probably have to modify my understanding on how this entire thing works: the font contains the glyphs which makup the image. This might help me figure out how to modify it...

nicolai86, 2018-11-21:
okay, http://teripom.x0.com/ can load the font from `glcdfont.c` and it allows me to edit it too. Downloading the modified `glcdfont.c` allows me to update the everything :)

closed 2018-11-21 by nicolai86

joric, 2020-05-19:
Made this thing https://github.com/joric/qle