foostan/crkbd #5: corne-classic .svg files are wrong size

debugtux, 2018-11-12:
Problem: .svg files from the _master branch_ located in the _corne-classic/case_ folder are incorrect size (scaled down).
I've detected the issue in the corne-classic-tent.svg and corne-classic.svg files.
Resulting in too tiny case plates from the lasercutter when using the .svg files included.

Note: The .ai files do look to be the right scale, however my crkbd kit is still in transit therefore I am unable to verify their scale correctness. They look good to the eye, where the .svg certainly do not.

Image below will illustrate the issue, on the left the .svg file on the right the .ai file (both loaded without modification into Inkscape)
[image]

foostan, 2018-11-13:
Thanks for your feedback. I'll fix svg files.

AdriaFa, 2022-09-23:
Where I can find this svg files? Thanks you!! (: