foostan/crkbd #6: tenting solutions

nicolai86, 2018-11-20:
This is more a question than anything else: 

are there 3d printable tenting solutions for crkbd available? I started using it full-time at work today and that’s like the only thing I’m missing.

If there are crkbd tents maybe we can document them here?

leotumwattana, 2018-12-01:
Would love to know as well.

leotumwattana, 2018-12-01:
I asked foostan on Twitter and he pointed me to the tenting plates here: https://github.com/foostan/crkbd/tree/master/corne-classic/case

nicolai86, 2018-12-02:
That’s a good start; I’ll take a look if I can design a 3d printable tent for the regular plates tho as it would be way cheaper than getting new plates.

lukeberry99, 2021-05-14:
I hate to revive an old issue, but that link no longer works, has it been moved elsewhere?

keyboard-magpie, 2021-05-14:
https://github.com/foostan/crkbd/tree/master/plates/tilting-tenting

lukeberry99, 2021-05-14:
Thankyou @keyboard-magpie