foostan/crkbd #11: Kicad Gerber File Board outline not detected by JLCPCB

BenRoe, 2018-12-12:
Hello,
i created the Gerber files wit KiCad and uploaded it to JLCPCB, but the outlines are not detected.
[image]
[image]

This are my settings.
[image]
[image]

I tried different settings, but every time the same problem.
Any suggestions?

foostan, 2018-12-31:
Can output correct gerbers with those settings.
What version are you using?

[image]

[image]

BenRoe, 2018-12-31:
It works. Many thanks.
I use version 6.0.0-rc1-dev-1291

closed 2018-12-31 by BenRoe

fi0, 2019-10-22:
@BenRoe do you mind sharing the gerber files for crone classic please? I met similar issues on JLCPCB. thank you!

Zeioth, 2020-08-07:
I'm also ordering from JLCPCB. What size should I indicate?