foostan/crkbd #12: Wondering about QMK Proton C

leotumwattana, 2019-01-26:
The QMK Proton C seems like a great new controller with more flash and faster ARM based chip. Was wondering if anyone tried using it with Corne?

leotumwattana, 2019-01-30:
Yeah. I know about the Elite-C. I personally don't care that much about the connector. That's because I use magnetic connectors. So whether the connector is USB-C or Micro doesn't matter to me too much.

I'm more interested in the ARM based chip. Which has more flash so that I can turn on more features. For example, right now, I have to turn some QMK features off so that I can fit in Dynamic Macros. Would be nice if I can basically turn everything on. 😆 

I also don't mind breaking off extra pins from the Proton C. I would rather keep the beautiful profile of the Corne over having sound (personal preference). Of course, if @foostan figures something out that looks nice, I'm game for it.

jeffreykxiao, 2019-01-31:
@leotumwattana The current version of the QMK firmware has no support for ARM-based chips on split keyboards. You can check out `split_common` in the QMK firmware and you'll see everything there is for AVR. Unfortunately, I don't think that @foostan can really do anything about this; IIRC the effort for getting ARM support for splits hasn't started yet and it might be a long way off. I'd like to use the Proton C myself, but at this point it's either wait or contribute to QMK.

leotumwattana, 2019-01-31:
@jeffreykxiao Ah, ic. Thanks for explaining the situation. I guess it's a waiting game for now. Look forward to it though.

snowe2010, 2019-04-09:
@kmanwar89 do you have some pictures and maybe a tutorial of your Elite-C on your crkbd? I would like to use usb-c with my incoming crkbd if possible.

kmanwar89, 2019-04-18:
> 
> 
> @kmanwar89 do you have some pictures and maybe a tutorial of your Elite-C on your crkbd? I would like to use usb-c with my incoming crkbd if possible.

Apologies @snowe2010, the notification for this comment went to an email I almost never check.  I can take some pictures and unicast them to you when I get a free chance this evening/this weekend?  I honestly built the board and don't use it, but that's mostly because I'm putting off having to program it and learn QMK....

I will say I really enjoy the USB-C connector, since most phones have USB-C now and the cables are far more prevalent.

snowe2010, 2019-04-18:
That would be awesome. Thanks!

danielo515, 2019-05-03:
> That's because I use magnetic connectors. So whether the connector is USB-C or Micro doesn't matter to me too much.

Would you mind sharing your magnetic conector?

leotumwattana, 2019-05-03:
Sure. I use this one from XPower: http://www.xpower.com.hk/magnetic-cable

kmanwar89, 2019-06-02:
> 
> 
> That would be awesome. Thanks!

@snowe2010 I am the worst -- I told you I'd take a screenshot and never did. If you still need them, I can take some and upload them.  Honestly though, I just followed the build guide and just used an Elite-C instead of a Pro Micro. Wired and soldered it the exact same way, since it's the same board. The CRKBD doesn't have room for the extra breakout pins (which can be used to control a speaker, how cool!), so make sure you only buy the version without it (or break it off if you do get it on your controller).

Other than that, it's just a drop-in replacement for the ProMicro from my experience. This is my first ever board I've built and first time ever doing anything with soldering or small electronics, so take my advice with a heavy dab of salt!

snowe2010, 2019-06-03:
@kmanwar89 	ok, no worries. As long as it's drop in it should be fine.

kmanwar89, 2019-06-10:
@snowe2010  yup. drop-in replacement for Pro Micro.

joric, 2019-08-04:
You could use this https://github.com/joric/nrfmicro/wiki/Corne-BLE

closed 2019-09-30 by foostan