foostan/crkbd #13: what are the keycaps in the photos for the chocolate switches?

pvinis, 2019-02-12:


jeffreykxiao, 2019-02-12:
Designed by a user named `monksoffunk` on the Japanese Keyboard Discord

Link to product: https://make.dmm.com/item/1012817/

No idea if there's sale outside of Japan, maybe best to contact him. No idea about whether or not he speaks English.

pvinis, 2019-02-12:
Thanks for the info. I'll see what I can do to get them! :)

closed 2019-02-12 by pvinis

rjhilgefort, 2019-12-03:
@pvinis did you find these caps? Looking for them as well

pvinis, 2019-12-03:
Last time I talked to (u/monksoffunk)[https://www.reddit.com/user/monksoffunk] and basically he sells (here)[https://make.dmm.com/item/1012817/] and he might sell (here)[https://25keys.booth.pm/] in the future for EU.

rjhilgefort, 2019-12-04:
Cool, yeah, I just can't figure out how to order them even with google translate on. Thanks for the reply!

pvinis, 2019-12-05:
I haven't tried yet. Maybe let's talk together and try to figure it out! Find me on anything else like twitter or discord, `pvinis` everywhere.

rjhilgefort, 2019-12-05:
DM'd!

IncognitoFawkes, 2024-05-30:
Were you guys able to figure out how to get the keycaps?