foostan/crkbd #17: Purpose of extra breakout pins

filoxo, 2019-05-05:
Hi! I had a question about the extra pins that are broken out just below the pins for the OLED screen, labelled P1, P2, R1, and R2. 

[image]

Typically I would expect R to mean resistor. I have no idea what P could mean here. 

I've typically seen the need for 2 resistors to enable [i<sup>2</sup>c](https://github.com/qmk/qmk_firmware/tree/master/keyboards/lets_split#wiring) which I have not seen that mentioned in any of the documentation; but [i<sup>2</sup>c seems to be enabled by default in QMK](https://github.com/qmk/qmk_firmware/blob/master/keyboards/crkbd/rules.mk#L70) for this keyboard. Any info would be appreciated! Thank you in advance!

filoxo, 2019-05-05:
Was examining the traces, and the P holes seem to just be breakouts for those wires, nothing more, and they lead to the TRRS jack. The Rs are for resistors, and that indicates to me that these are for enabling i2c proper. Interesting that the build guide does not mention these and i2c is enabled by default. It should probably be defined to use serial instead. I wonder if this is the reason I'm having issues with the RGB_MATRIX stuff right now.

foostan, 2019-06-03:
> Was examining the traces, and the P holes seem to just be breakouts for those wires

That's right!

> Interesting that the build guide does not mention these and i2c is enabled by default.

It's for i2c but the firmware of  crkbd is not compatible yet.

> It should probably be defined to use serial instead. 

The crkbd use serial by default, but i2c used by OLED display.

Thanks.

filoxo, 2019-06-03:
Thanks for the info!

closed 2019-06-03 by filoxo

dsanchezseco, 2019-08-21:
Hi!! I follow this road to and ended here(sorry for the message in the closed "issue")

So I guess that breaking the data trace and adding the 4.7k resistors in R1-P1 and R2-P2 would do the trick to enable I2C in the board although the software is not ready.

Are you working on it? I maybe give it a try as I was working on my keymap and saw that the oled config is outdated and there's a new lib for it.

just two quick questions, y suppose that the data trace has to be broken only in one half as well as only two resistors are required, right?

also thank for this board, is the second day I'm  using it and I love it!!

grant-park, 2019-12-23:
Hi @dsanchezseco 

Can you share a picture of the part of your PCB that has the changes necessary for i2c? I have no idea how to follow steps 1-3 without risking ruining my PCB, especially since you mentioned breaking the data trace.

dsanchezseco, 2019-12-29:
Hi! @grant-park i didn't do it, i was just asking as you how to do it and thinking about the procedure looking at other keebs and the qmk docs. Sorry