foostan/crkbd #18: TRS Cable is Insufficient

cgrange, 2019-05-10:
The build guide says that TRS or TRRS cables can be used. I tried mine initially with a TRS cable and whatever side was plugged into the usb cable worked but the other would not. Once I replaced the TRS cable with a TRRS cable both halves worked perfectly. I could be the exception to the rule but I doubt it since most split keyboards require TRRS. I wonder if the build guides are wrong in stating that either TRS or TRRS cables will work.

foostan, 2019-06-03:
The Crkbd can use TRS cable.

Please check kicad data.
[image]

vxid, 2019-07-01:
I had the same issue as @cgrange. TRS did not work, TRRS did.

gelpicr, 2019-07-13:
I actually had the same issue with my build, but I was able to get it running by plugging in the TRS cable then connecting both sides to USB. It's obviously not ideal, but i'm still waiting on my TRRS cable to come in the mail. Glad to know I wasn't the only one and that TRRS should fix my problem as well.

note: I'm using the cherry hotswap version rather than the classic corne if that makes any difference.

tominabox1, 2019-08-21:
I have also tried using a TRS cable with no success - Power was provided but appeared to be no data.

foostan, 2019-09-30:
How long your TRS cable?
I saw an issue where left and right communication fails when using a cable that is too long.

tominabox1, 2019-09-30:
I'm using rev 1.1 boards if that has anything to do with it. The cable I
used is only 3 feet I think.

foostan, 2019-10-02:
https://github.com/qmk/qmk_firmware/blob/master/keyboards/crkbd/serial.c#L137

Please try changing `SELECT_SOFT_SERIAL_SPEED` value.

tominabox1, 2019-10-03:
Trying on my dual pro-micro board now and it is actually working with a short cable... I'll have to check on my other one!  I didn't change any config options.

tominabox1, 2019-10-14:
Confirmed TRS works on both of my cornes without changing any settings. I believe I had a bad solder joint on the one board I was having issues with.

foostan, 2019-10-18:
I’m glad working your corne. Thanks.

closed 2019-10-18 by foostan

BeaRRRRR, 2020-03-06:
I got my hotswap cherry corne working with a 3 meter TRS cable, so I think that cable length isn't really important.

Brammm, 2020-09-08:
@cgrange did you ever get TRS working? I have a high quality Rode TRS cable but experiencing the same issue.

vguerci, 2020-09-08:
After diagnosing a similar issue, I can confirm that `TRS` should work. Thus there is an issue that happens easily, if you short the TRRS Socket pads there: [image]

When A-D are shorted, this is not an issue with TRRS cables but is one with TRS cables that short B-D pads. VCC and data are then shorted.

I hope that helps.

Brammm, 2020-09-08:
Hmm, I'm using a Corne Cherry PCB but with some slight modifications (among which to use RGB LEDs with legs that solder more easily) and on mine, those 4 connections of the TRRS jack are in one line... I'll get in touch with the dude that modified the PCB's.

vguerci, 2020-09-08:
The Worldspawn's "easy LED" version I guess. This one short by design the A-D pads and cannot work with TRS cables.