foostan/crkbd #19: Corne Classic v2.0 OLED Power?

taylorcoffelt, 2019-06-02:
I have the v2.0 boards of the Corne Classic, and I can't seem to get the VCC pin for the OLED to have power. 
I have the top OLED jumpers soldered on both sides. I'm currently using the Elite-C, and the test keypresses do work. Following the traces on the board, KiCad design, and schematic, it _should_ have power. 
 Anyone else have this issue?

foostan, 2019-06-03:
> I have the top OLED jumpers soldered on both sides

Maybe it's mistake. You should solder one side. 
Please check [buildguide](https://github.com/foostan/crkbd/blob/master/corne-classic/doc/buildguide_en.md#jumpers-for-oled-modules-optional).

taylorcoffelt, 2019-06-10:
My apologies, I was unclear in that statement.
I have soldered the OLED Jumpers on the top side of the PCB for both left and right boards, according to the instructions in the buildguide. I'm not seeing anything that is different in my setup, besides the fact that the buildguide is for v1.1 of the board, and mine is v2.0

foostan, 2019-09-30:
> Anyone else have this issue?

No, I have not see this issue.
Please show me your assembly.

closed 2020-04-18 by foostan