foostan/crkbd #20: Building without OLED modules, can I add later?

ollyhayes, 2019-06-26:
Hi,

I'm currently building the keyboard and don't have any OLED modules yet, I'd like to add them later if possible.
About short circuit the OLED jumpers, I'm not sure I'm going to be able to get to them once the pro micro is soldered into place... What happens if I just solder them now? Will the keyboard work without the OLED modules attached?

Thanks

juno, 2019-06-27:
In short, it works without OLED modules and you can attach it later.

Initially, I use the Corne with soldered jumpers and OLED modules but later I noticed that I don't use any OLED features. Then, I removed the modules and no any probs.

ollyhayes, 2019-06-28:
Ah great, thanks very much. I'll solder them all now then.

closed 2019-06-28 by ollyhayes