foostan/crkbd #22: How to agencies tenting plates?

atomicwrites, 2019-09-07:
Hi, I wanted to use the tenting plates but I can't find any information about them, how they are assembled, what thickness to cut them in, what all the washers in the design file are for, or what other hardware I need. I'm guessing your supposed to put some kind of bolt in the holes on the side and do you use normal standoffs to attach the PCB? Thanks.

atomicwrites, 2019-09-07:
Oops, autocorrect got me on the title and I can't edit it, that should say "assemble."

foostan, 2019-09-30:
Dit you set https://github.com/foostan/crkbd/blob/master/corne-cherry/doc/buildguide_tilting_tenting_plate_jp.md ?

I uploaded case files already.
https://github.com/foostan/crkbd/tree/master/corne-classic/acrylic_plate

closed 2019-10-02 by foostan