foostan/crkbd #32: Tilting tenting plate confusion

markau, 2019-10-14:
The tilting tenting [build guide](https://github.com/foostan/crkbd/blob/master/corne-cherry/doc/buildguide_tilting_tenting_plate_jp.md) suggests I basically need 2 top plates... one from [the tenting file](https://github.com/foostan/crkbd/tree/master/case/acrylic_plate/tilting-tenting), and the top plate from the regular [cherry case](https://github.com/foostan/crkbd/tree/master/case/acrylic_plate/cherry)).

Is this correct?

I was going to have the tenting plates cut from 3mm acrylic. The build guide says for the top plate `Corne Chocolate / 3mm top plate is not supported`.  So if I need 2 top plates, how thick should the regular plate be?

thoo, 2019-11-08:
https://github.com/foostan/crkbd/tree/master/case/acrylic_plate/tilting-tenting
This is both top and bottom plate.

foostan, 2019-11-14:
Use the corne cherry kit that includes PCB top plate and [the tenting file](https://github.com/foostan/crkbd/tree/master/case/acrylic_plate/tilting-tenting) in [The build guide](https://github.com/foostan/crkbd/blob/master/corne-cherry/doc/buildguide_tilting_tenting_plate_jp.md).

Do you want tilting with Corne Chocolate?
You should 2mm thickness plate only acrylic. Cannot with PCB plate(it is basically 1.6 mm thickness).

closed 2020-04-18 by foostan