foostan/crkbd #33: Tilting pcb top & bottom

pkieszcz, 2019-10-17:
@foostan is there any chance that you will provide a files for tented version of top & bottom PCB?

foostan, 2019-11-14:
Thanks for your feedback, I will consider.

Q-Storm, 2019-12-09:
That would be awesome if you could provide this.

hoijui, 2020-07-06:
yeah that would be cool!

As I imagine it, the easiest way would be, to take the tilting case SVGs and convert them to a KiCAD PCB.
It seems like [this tool](http://mondalaci.github.io/dxf2svg2kicad/) can do that.
The code it uses to do that [can be found here (JavaScript, GPLv3)](https://github.com/mondalaci/dxf2svg2kicad/blob/master/scripts/svgToKicadPcb.js).

hoijui, 2020-07-06:
I tried it, and either I do not understand how this web thing works, or.. it does not work (I can not get the generated KiCad file).
I guess some JS wizard would have to have a look at it.

hoijui, 2020-07-29:
... yet an other option - which seems more feasible to me right now - would be to convert he SVG (generated from the AI files) to DXF, which can then be imported into KiCad. this is something I already did with an SVG logo, so that should work.

I used [svgToDxf](https://github.com/winder/svgToDxf), but there might be better options out there.

closed 2024-11-15 by foostan