foostan/crkbd #36: Corne classic, Alps and switch orientation.

oprietop, 2019-11-09:
Is there any chance to consider rotating the switches in the corne classic?  
After building some boards with SKCM Alps they definitelly feel and behave differently with the switchplate facing south. Original Alps switches tend to be placed with the switchplate facing north in their original boards and they do feel feel mushier and even scratchy when rotated. 
To be fair i guess this won't be a problem with NOS Alps switches but it's almost impossible to have this kind of switches in this state today, the common practice is to get switches from older, north-facing switchplate boards.
I also know rotated switches can be troublesome with certain GMK sets so that could be also a plus.

foostan, 2019-11-14:
I know problem for north facing recently. I will consider.

jeffreykxiao, 2019-11-14:
Hm - in Apple M0116 and II GS keyboards the switches for rows 1-3 are all rotated but I can't say I've ever noticed the difference on those boards. But for MX + GMK I definitely agree.

equalunique, 2019-11-23:
I have Corne Classic PCBs and was planning to do Alps builds with them. Thank you for sharing this warning.

rpbaptist, 2020-01-24:
It's noticable with R3 and R4 Cherry profile caps as well. Especially on R3.

foostan, 2026-11-14:
Thanks for the suggestion and the discussion, and sorry for leaving this open for so long. Unfortunately, older versions such as Corne Classic are no longer supported, so I won't be making this change.

I'll close this issue. Thank you for your understanding.

closed 2026-11-14 by foostan