foostan/crkbd #41: Support for Kailh 1232 series

an-tex, 2020-01-13:
Kailh released new low profile switches with an even lower total and pre travel. Unfortunately they also have a different form factor for the base. 

I'm not sure they can be used in combination with the hot swap sockets but it doesn't look like.

Announcement: https://www.kailhswitch.com/mechanical-keyboard-switches/low-profile-key-switches/dip-brown-key-switches.html
Spec: https://www.kailhswitch.com/uploads/201815927/CPG123201D01---SPEC.pdf
Buy: https://www.aliexpress.com/item/4000277394324.html?spm=a2g0o.cart.0.0.5a643c00zC73rZ&mp=1

The brown switch looks amazing, almost no pretravel. Seems to be the low profile version of the Kailh Bronze.

foostan, 2020-01-28:
Thank you for your question.
Corne classic and Corne chocolate can use kailh low profile switches.
Could you confirm it?

an-tex, 2020-01-28:
Hi @foostan ,
thanks for your reply! I'm afraid not. These "Low profile Kailh 1232 series" are different to the "Low profile Kailh PG1350 Choc series"
The cutout at the bottom is not just a circle, but has also some rectangles. The contacts are placed differently as well.

Here are two photos for comparison:

- [PG1232](https://ae01.alicdn.com/kf/H37b6b16d47904ac98cfb3b165a495807M/Kailh-Low-profile-Mechanical-Keyboard-Switch-ultrathin-keyboard-switch-for-laptop-linear-tactile-handfeelling-wholesale-CPG1232.jpg)
- [PG1350 (Choc)](https://ae01.alicdn.com/kf/H7c2cec51edb14dd3979a312d977c94e7g/Kailh-low-profile-Switch-1350-Chocolate-Keyboard-Switch-RGB-SMD-kailh-Mechanical-Keyboard-white-stem-clicky.jpg)

foostan, 2020-01-29:
Wow, "Low profile Kailh 1232 series" is are new switches!
OK, thanks confirming. I will check it.

foostan, 2021-10-30:
This switch is not general, so I decided not to use it.

closed 2021-10-30 by foostan

ec1oud, 2023-01-25:
But I found https://github.com/davidphilipbarr/36keys/tree/master/42Keys/chocorne%20low