foostan/crkbd #42: Design Review: Recessed RGB

NellyWhadsDev, 2020-02-16:
Hey there,

I'm working on a keyboard of my own and was wondering what your motivation was for recessing the RGBs into the board, as opposed mounting them on the surface. Was it mainly for building ease, so that builders don't have to fuss with actual SMD work, or was there another reason?

Also, if the LEDs are recessed into the board, does it affect their light distribution into the switch itself? I just can't seem to find a side-by-side comparison of how the light disperses into the switch on this board vs a non-recessed one (using the same camera, angle. etc.)

Thanks! And awesome keyboard you have here!

BenRoe, 2020-02-18:
I guess it‘s because if the per-key leds are surface mounted the switch wouldn’t fit.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

NellyWhadsDev, 2020-02-21:
There are Keyboards with surface mounted LEDs which do fit underneath keys. Perhaps it was to support the different led position between MX and Choc switches?

foostan, 2020-02-22:
Crkbd forked from [Helix keyboard](https://github.com/MakotoKurauchi/helix) that has per-key LEDs(SK6812MINI). SK6812MINI is control RGB and brighting simply with serial cascade without other controller. But I and most of builder think assemble parts so difficult.

Do you know small LED that can control RGB and brightening without another controller?

foostan, 2020-02-22:
Maybe I will use [YS-SK6812MINI-E](https://yushakobo.jp/shop/ys-sk6812mini-e/) if re-design crkbd. It's like SK6812MINI and assembling to be easy.

NellyWhadsDev, 2020-02-23:
There seems to be some confusion on my end; My understanding was that most keyboards had LEDs installed on top of the board, like this: https://www.1upkeyboards.com/shop/controllers/1up-rgb-pcb-hse/

But upon further digging, it seems most of these 6812s are actually installed into the PCB when it comes to per-key implementations. My mistake.

BenRoe, 2020-02-23:
YS-SK6812MINI-E look promising. @foostan do they have a different pinout or why they are not compatible with the current pcb?

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

foostan, 2020-03-03:
YS-SK6812MINI-E is a not compatible with SK6812MINI.

closed 2020-03-03 by foostan