foostan/crkbd #46: How can I diy Logo easy?

liangzhaoliang95, 2020-04-12:
How can I diy Logo easy?

benoittgt, 2020-04-12:
You mean the logo on the PCB? or LCD?

liangzhaoliang95, 2020-04-12:
LCD LOGO, I hava solved it now ,but i think qmk should provide a simple tool to change the logo.

benoittgt, 2020-04-12:
Out of thread: I did similar changes on ergodox: https://github.com/benoittgt/controller/tree/ergodox-miami-theme

[image]

liangzhaoliang95, 2020-04-12:
[image]
my keybord is crkbd

benoittgt, 2020-04-12:
Congrats. It's beautiful! 🙌🏻

foostan, 2020-04-18:
Oh, sorry for late return message.
@benoittgt Thank you for helping 🙏

closed 2020-04-18 by foostan

joric, 2020-05-19:
@liangzhaoliang95 https://github.com/joric/qle