foostan/crkbd #49: Support for Kailh Choc V2 Switches

connorbowden, 2020-06-01:
Kailh has released a second version of low profile switches, called Choc V2’s. From what I’ve read, the pin layout is different from both the other version of choc low profile switches and cherry’s low profile switch, which would require a new PCB layout. Will there be an update for Corne’s PCB to support these switches?

filoxo, 2020-06-11:
Duplicate of #41

connorbowden, 2020-06-12:
@filoxo Link to the first version release of Kailh's low profile switches: https://www.kailhswitch.com/mechanical-keyboard-switches/low-profile-key-switches/dark-yellow-switch.html

Link to the 2nd version of Kailh's low profile switches (the ones I referenced) : https://www.kailhswitch.com/news/kailh-choc-v-32057417.html

foostan, 2020-07-29:
I'm try prototyping for Choc V2. 
[image]

MarkCampbell1970, 2020-10-29:
Hello, Did you have any luck with prototyping PCB. I actually bought some of the choc v2 switches thinking that they would fit on the regular Kailh PCB for low profile switches and obviously that didn't work out. I am not able to find any other board that supports it. I was just wondering if you still planning to making a prototype and if you plan to make it public.
Thank you!

CMorooney, 2021-04-28:
> Hello, Did you have any luck with prototyping PCB. I actually bought some of the choc v2 switches thinking that they would fit on the regular Kailh PCB for low profile switches and obviously that [...]

Same here -- crkbd is my favorite layout/board and I am really looking forward to putting these switches on one! Any news on official compatibility?

keyboard-magpie, 2021-04-28:
Corne light in the main repository has support for choc v2 now @CMorooney https://github.com/foostan/crkbd/tree/master/corne-light

CMorooney, 2021-04-28:
ah, great, thanks @keyboard-magpie ! Perhaps this issue should be closed?

markstos, 2021-04-28:
I'm building a Corne Light with Choc v2 switches now. It's currently with a builder who is debugging some issue. I got my PCB from a Redditor who had several of the prototypes printed for himself and sold one to me. I don't think he has more. (And was located in Sweden, so the shipping cost to the US was about the cost of the PCB).

closed 2021-10-30 by foostan