foostan/crkbd #56: RGB backlight not working

JJoe21, 2020-07-14:
Backlight doesn't work while underglow does. Did i burn the leds?

foostan, 2020-07-29:
@JJoe21 This is not a place for troubleshooting. Would you send e-mail to me with the detail of problem and some pictures.

closed 2020-07-29 by foostan

MadRabbit, 2020-10-05:
i have the same issue... been fiddling with my PCB for most of the day with no avail. unfortunately there is not much info online on how to troubleshoot the problem. im struggling to turn all of the LEDs on as per doc. i basically did what the QMK readme says for crkbd, but that only got me the underglow working, not the backlight. and its for both halfs, so im suspecting there is something else going besides soldering.

if you could share some of your wizdom on this, that would be awesome! a wiki page maybe? or a paragraph in the build docs to explain how to turn the backlight LEDs on?

pretty please?

keyboard-magpie, 2020-10-05:
As foostan said, this isn't a place for tech support. 

Discord, however, is a much better place: https://discord.gg/qNtnWp