foostan/crkbd #58: RGB footprint pin marking is on 4 rather than 1

DaneEvans, 2020-08-07:
Standard engineering would have the marking on the footprint on pin 1 - 

this works if you are using the build logs, but if you know what you are doing, and go to the data sheets, you end up realising that the geometry just doesn't work. Then you go looking at the build documents, and realise that for some reason convention was thrown out the window, and the pin 1 marking is actually on pin 4 - Vdd 

Changing it now is probably not worth the additional headaches, and as I said, the geometry doesn't work, so it's unlikely to fry LEDs, this is more here in case someone else goes looking for an answer.

WernerFS, 2020-10-15:
I have been working in a fork version of this same keyboard and doing the BOM I discovered this. I almost lost my mind when I checked the documentation. I will correct the footprints when I have time and will upload in a personal fork :) this is an amazing keyboard that has a lot of potential.

WernerFS, 2020-10-15:
> I have been working in a fork version of this same keyboard and doing the BOM I discovered this. I almost lost my mind when I checked the documentation. I will correct the footprints when I have time and will upload in a personal fork :) this is an amazing keyboard that has a lot of potential.

Disregard this, I realize there are several footprints and the one in this datasheet **https://cdn-shop.adafruit.com/product-files/2686/SK6812MINI_REV.01-1-2.pdf** so I guess you should be careful when sourcing your components haha :D

DaneEvans, 2020-10-15:
That might explain the confusion, though that datasheet is an abomination.

The pin 1 index is visible on the diagram of the LED, yet that pin is
labelled pin 3.

This it the problem with knockoffs.
An alternative might be to add the notch on the silkscreen, as it appears
to be consistent.

agusdmb, 2022-10-12:
this is the same as #166

DaneEvans, 2022-10-13:
No it's not

agusdmb, 2022-10-13:
> No it's not
> […](#)

:man_facepalming: you are right... i miss read this somehow... now i just read it again and realized it.. i deleted my comment to avoid having the links between the 2 separated issues

DaneEvans, 2022-10-13:
All good.
I figured it was something of the sort

closed 2023-11-11 by foostan