foostan/crkbd #65: There are errors in the documentation.

Coro365, 2020-10-02:
I think there is an error in the following documents.

[crkbd/buildguide_jp.md at master · foostan/crkbd](https://github.com/foostan/crkbd/blob/master/corne-cherry/doc/v2/buildguide_jp.md#%E3%83%95%E3%82%A1%E3%83%BC%E3%83%A0%E3%82%A6%E3%82%A7%E3%82%A2)

If the command is described in the documentation, it will result in an error.
```
make crkbd:default
make: *** No rule to make target `crkbd:default'.  Stop.
```
I followed the following documentation and it worked.
[セットアップ](https://docs.qmk.fm/#/ja/newbs_getting_started?id=_4-%e3%83%93%e3%83%ab%e3%83%89%e7%92%b0%e5%a2%83%e3%81%ae%e7%a2%ba%e8%aa%8d)

```
qmk compile -kb crkbd -km default
Ψ Compiling keymap with make crkbd:default

QMK Firmware 0.10.24
Making crkbd/rev1/legacy with keymap default

avr-gcc (Homebrew AVR GCC 8.4.0) 8.4.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

```
Is this the right approach?
Thank you!

con-f-use, 2020-10-08:
The make command if for when you have cloned the qmk repository from github yourself and are cded into it. So this is not an error.

Coro365, 2020-10-19:
Thanks to you and people like you, I've been able to write successfully!
Thank you.

```
qmk compile -kb crkbd -km default
qmk flash -kb crkbd -km default
```

closed 2020-10-19 by Coro365