foostan/crkbd #66: Problem with OLED displays

con-f-use, 2020-10-08:
I've build two cornes and have a problem with the OLED displays.

They don't show anything, on both keyboars (all 4 halves). I've tried a swapping them out and switching OLED displays between them. Some of displays are even from a different seller. I noticed, that when I don't have them in, the keyboard reacts a lot slower. All this leads me to the conclusion, that the OLEDs are probably not deffect, and I've done something wrong. 

Do you have any troubleshooting advice? Is there a special firmware option, I have to set in QMK to enable the OLED?

keyboard-magpie, 2020-10-08:
Did you solder the jumpers on the top of the PCB?

keyboard-magpie, 2020-10-08:
https://discord.gg/ApVVt6 come join foostan's server for a chat through the issues.

con-f-use, 2020-10-12:
The issue was resolved on the discord servers. The current workaround is to compile the firmware with this fork of qmk . Cause seems to be that the used oled library is incompatible with newer oled modules.

See: 
 - https://github.com/qmk/qmk_firmware/pull/10354;
 - https://discordapp.com/channels/715899960327012423/742781449731702824/763820765216112650

foostan, 2020-11-13:
Please see https://github.com/foostan/crkbd/blob/master/doc/firmware_en.md.

closed 2020-11-13 by foostan