foostan/crkbd #67: CRKBD doesn't work

mralkofgihter, 2020-10-13:
Hello, have an issue with my corne keyboard.
I've got cherry v2.1 pcbs.

I soldered first PCB, flashed my pro micro board and the problem is, that pro micro doesn't work when plugged to pcb.

I've checked all the  soldered joints, diodes, but can't find what's the cause of it.

Maybe someone got the same problem? Help is strongly needed.

Today I'm going to solder the seconds pcb to check if the problem will appear again.

Thank you.

keyboard-magpie, 2020-10-13:
Come join the pastry keyboard discord server (foostan's server) and we can talk through issues and see pics.

https://discord.gg/8tmn7p

foostan, 2020-10-21:
Thank you for building and your information.
Assembly problems are supported by the discord server.

closed 2020-10-21 by foostan