foostan/crkbd #70: feature request: rotary encoders

beraldoleal, 2020-10-24:
Hi, first of all, would like to say this is a nice work. I really like this design. I have one question: How difficult would be to add a rotary encoder in this layout (specially on the light version)? (maybe moving the OLED a little high and adding below it, not sure).

jack-davidson, 2020-11-30:
I don't think it would be that difficult. Check out the designs for some of the keyboards which do support rotary encoders and see how this is done in the pcb. I think you would have to do some qmk configuration but that is in the qmk documentation.

A better idea may be to start a keyboard project derived from the crkbd but add a place to put a rotary encoder. I don't know if the maintainers of this project like rotary encoders as the crkbd is a very compact design and you might have to sacrifice a key for a rotary encoder if you wanted your changes to be accepted back into upstream.

paulhersch, 2020-12-06:
I think a rotary encoder would be best at the main thumb key, as it allows to scroll through text, e.g. if you use vim and want to get to a character or if you are reading an article and want to scroll down without touching your mouse.

I really need to learn how to design PCBs lol

jack-davidson, 2020-12-06:
Possibly the maintainers would allow rotaryencoder version of the crkbd (just like the chocolate, cherry, etc) and it could be another version of the crkbd. If you need to scroll down in an article you can probably use browser keybinds for that and in vim you can use w and b for moving around characters. That isn't to say that a rotary encoder is not useful, just saying you can use those methods in the interim period between getting a rotary encoder on a crkbd.

WernerFS, 2021-01-09:
Hi, I am currently working on a modified version with rotary encoder on one half and a joystick on the other half. I must say I also got rid off the Pro Micro and substituted with a LORA board from TTGO. This board supposedly can handle Bluetooth and Wifi. I will let add the project to my github once I get into testing. For now I only have the boards.

jack-davidson, 2021-01-09:
What is the normally-supported ble board for use with qmk? I found this
article on the qmk docs
https://beta.docs.qmk.fm/using-qmk/hardware-features/feature_bluetooth and
I think that may be a better choice because it will be fully supported with
qmk and will maintain the same form factor. Was there another reason to get
rid of the pro micro?

WernerFS, 2021-01-09:
The main reason was to test the ESP32 (I got one for free) the extra connectivity already on board and the battery management. Sincerely it was mostly because I could use it as an excuse to experience with the ESP and LORA. The final product should be able to use the keyboard to chat via LORA.

jack-davidson, 2021-01-09:
Nice!

foostan, 2021-10-30:
First of all, I want to keep the crkbd design simple. I also think keeping backward compatibility is important.

In other words, I don't intend to change the location of the keys or place new parts in another location. If there is such a desire, I recommend forking and designing a new one.

However, I would like to be able to place the rotary encoder in place of the existing keys. I'll consider it in the next version.

foostan, 2024-03-16:
https://github.com/foostan/crkbd/releases

closed 2024-03-16 by foostan

beraldoleal, 2025-10-22:
Hey @foostan, I just saw your latest update. I’ve been a bit busy, but I really appreciate it.

Again, awesome work.