foostan/crkbd #95: Corne Chocolate Top Plate drill layer vs outline issue

yourfate, 2021-03-05:
I'm about to have PCBs and plates for the corne chocolate produced by JLC, and I sent them the gerber files from the release page here: 

https://github.com/foostan/crkbd/releases/tag/corne-chocolate-v2.1.0

Now I got the reply that there are cutouts in the drill layer that get cut off by the outline layer, see JLC email here:

```
Hi Sir,

Well got your order with many thanks~

Sorry to bother you, but there is an issue that we want to confirm with you before proceeding.

As shown below, the slots on the drill layer will be cut off by the slots on the board outline layer,

is it normal to cut them out per the board outline layer as non-plated slots?

Your early reply will be highly appreciated!

Thank you so much!

```

[image]

martinmunk, 2021-09-22:
Hi @yourfate 
I'm considering trying my luck with making a Choch version as well, and maybe also use JLC.
Did you end up getting the issue resolved?

yourfate, 2021-09-22:
Hi @martinmunk, sorry for now following up on this issue.

Yes, they delivered the plates and they worked fine. You can see the result here:
[image]

I would recommend using thinner PCB for the plates than I did, as 1.6mm is too thick for the switches' clamps to reach below. It works, but it would be neater using 1.2mm.

Side note: You can specify "Remove Order Number" at JLC, I forgot to do that, and now have order numbers on my plates and PCBs.

closed 2021-09-22 by yourfate

pavel-rakhmanov, 2021-11-26:
Also faced with this issue while ordering corne-chocolate-v2.1.0 plates from JLC PCB but with a slightly different wording:

> As shown below, for all square slots we can only make them non-plated, is it ok？

[image]
[image]

saeedb3r, 2022-12-17:
Hello everyone
I am currently in the process of building a corn chocolate keyboard, and I have received a message from the PCB manufacturer regarding the top layer. I apologize for my lack of knowledge on the subject, but I was wondering if you could kindly help me understand what is required and how to respond to the manufacturer's message. Your expertise and guidance would be greatly appreciated.

-----------------------------------
our engineer has reviewed your file and order detail, here are questions to confirm :
1.For the square slot, should we produce as plated-slot? 
2.For the slot size of drill layer, which is inconsistent with the Edge-cut layer, should we produce according to Edge-cut layer  or drill layer ?
-----------------------------------

[image]