foostan/crkbd #98: Build-Guide versions-"mess" - cleanup?

hoijui, 2021-04-17:
```bash
$ find . -name "buildguide*.md"
./corne-light/doc/buildguide_jp.md
./corne-light/doc/v1/buildguide_jp.md
./corne-light/doc/v2/buildguide_low_edition_jp.md
./corne-cherry/doc/v3/buildguide_jp.md
./corne-cherry/doc/buildguide_jp.md
./corne-cherry/doc/v2/buildguide_jp.md
./corne-cherry/doc/v2/buildguide_tilting_tenting_plate_jp.md
./corne-classic/doc/buildguide_jp.md
./corne-classic/doc/buildguide_en.md
./corne-chocolate/doc/buildguide_jp.md
./corne-chocolate/doc/buildguide_en.md
```
Could/should this change?

`git` surely allows to handle this more elegantly, by using one branch per version, instead.
The question is, why you/whoever chose to do it this way - versionning by file-system, not by git branches,
and whether it should stay this way.

I would like to have a discussion here, as I want to understand the decision. I am not saying: "THIS IS RONG, YOU NEED OT HCANGE IT$%@" ;-)
I am also interested because we are developing standards for open source hardware development (in germany). I would be happy to get input!

WernerFS, 2021-04-23:
This is an interesting issue. I guess the fundamental problem is that the versions in this project don't necessarily mean linear development but more like parallel projectors . You could check this links: 

[Multiple branching techniques ](https://medium.com/@patrickporto/4-branching-workflows-for-git-30d0aaee7bf) 

[Parallel projects in one git repo](https://stackoverflow.com/questions/14679614/whats-the-best-practice-for-putting-multiple-projects-in-a-git-repository/45504910) 

I hope this helps. IMHO the kicad project could be refactorized with a central schematic and different pcbs inside the same project. Anyways versions are mostly regarding changes in pcb components or layout. 

I hope this is useful.

hoijui, 2021-04-25:
> the fundamental problem is that the versions in this project don't necessarily mean linear development but more like parallel projectors

Yes!
I think so too; parallel projects. ... But not fully!
It really are versions, where supposedly v3 is better then v2, but because many people already use v2 hardware, and it can not just be updated to v3 like software can, with a click of a button, it makes sense to keep both (all) versions documentation maintained and improving. For many people, the most straight-forward approach to this is, as it is done in this repository: each (major) version has its own directory. From a git/IT-tech point of view, having one branch per major version instead of one directory, makes more sense.
People know how to deal with directories, but not all know how to deal with git branches.
Then again, when using branches, it gets much easier to copy improvements from one version to the other, and to compare the differences between the versions. It also hides some of the complexity from the casual visitor of the project.

foostan, 2021-07-10:
Sorry for the late reply.

I'm also a software developer, so I'm used to the culture of branching each versions. I also understand the benefits of doing so.

The only point is that, as you mentioned, a lot of people that non developer visit this repository. So I think it's easier to separate the files.

One suggestion is to separate the build guide into a completely separate site. This will benefit from the operation of splitting branches.

However, the biggest issue is whether or not I have time to do it.

foostan, 2022-04-17:
This issue has been open for a while, but I'm closing it for now. I will think about it again when I create a new version.

closed 2022-04-17 by foostan