foostan/crkbd #100: Difference between the versions

hoijui, 2021-04-25:
When looking at the build guides, I see that there are different versions of some of them (eg. v2 and v3). While I could have a look at the PCBs, to see the difference, I would rather like to have a short summary of the difference right in the build guide, where I can/have to choose between v2 and v3. Maybe something like:

> Choose v2 if you are fixing/modifying a corne built before May 2020, or if you have the old controller chips.
> Otherwise, use v3.

hoijui, 2021-05-15:
That is a practical guideline of how to go at it when you already have the board.
The question is more about, how to know what the differences are, if you do not yet have a board.
Sure, we can just assume v3 is better then v2, but even then it would still be interesting to know in what way it is better, and do I need different parts? maybe v3 parts are not available in my country, but v2 parts are, and so on.
I think it would make sense to write down a small summary of the differences.

keyboard-magpie, 2021-05-15:
I prepared a spreadsheet here, that some people have found useful: https://docs.google.com/spreadsheets/d/1JLh0yxME6ry-PGoPHEyElbthcyuL1NXQGK774J_t8rU/edit#gid=0

hoijui, 2021-05-15:
> I prepared a spreadsheet here, that some people have found useful: https://docs.google.com/spreadsheets/d/1JLh0yxME6ry-PGoPHEyElbthcyuL1NXQGK774J_t8rU/edit#gid=0

ahhh nice!! :-)
thank you!
can you explain what these two mean:
* Handed PCBs
* Snappable to 5 col
?

I would then try to integrate it into the Markdown documentation here.

keyboard-magpie, 2021-05-15:
Handed PCBs: No= reversible PCB for left and right. Yes= non reversible PCB, separate left and right PCBs

Snappable: PCB incorporates a breakable section that can remove the outer column of keys:

https://images-ext-1.discordapp.net/external/RfLh3AOh9cvQ5FQWP3NrLSE1MzxT01U-j_HVVG_ccKk/https/media.discordapp.net/attachments/532682459800797196/830840096646365295/Animated_GIF-downsized_large.gif

closed 2022-04-17 by foostan