foostan/crkbd #107: ./plates/common/*.svg files broken

ceceba223, 2021-06-09:
The .svg files for the plate are incomplete, meaning that the blue and red cut-lines aren't visible.
[Here](https://imgur.com/a/LXpfRNw) is a screenshot of the bottom.svg opend in Inkscape.
This is also an error with other .svg files in the repositorie.

keyboard-magpie, 2021-06-09:
https://imgur.com/a/JcsbuDI

If viewed in a CAD program they are fully visible (this is QCAD on mac, freecad etc produce a similar view).

ceceba223, 2021-06-09:
I tried opening the file in LibreCAD but got the same problem.

keyboard-magpie, 2021-06-09:
I should prob do a PR with some working DXFs...

ceceba223, 2021-06-09:
What DXFs there are only .svg and .ai files

keyboard-magpie, 2021-06-09:
I have some I've made locally.

keyboard-magpie, 2021-06-09:
https://github.com/keyboard-magpie/crkbd-plate-files

criscol64, 2021-06-17:
> The .svg files for the plate are incomplete, meaning that the blue and red cut-lines aren't visible.
> [Here](https://imgur.com/a/LXpfRNw) is a screenshot of the bottom.svg opend in Inkscape.
> This is also an error with other .svg files in the repositorie.

The issue is the width of the lines, what I did was select all and then go to the properties of the lines and increase the witdh.

foostan, 2021-07-10:
It seems good to change the width of line.

medoix, 2021-07-11:
Adjusting the line width seems to not work for me in Inkscape. 
I get the following selection when selecting all.
Inkscape also shows the outlines when attempting to import the .AI file but then it does not import correctly.
Can someone upload/link a corrected SVG version?
[image]

foostan, 2021-08-24:
Thank you your commitment! https://github.com/foostan/crkbd/pull/112
I'll close this issue.

closed 2021-08-24 by foostan