foostan/crkbd #108: Understanding the electronics

DLopezGo90, 2021-06-10:
Hi 

As far as I know both Pro Micro (left and right hand) communicate with each other using serial communication. For this I thought that RX from device 1 needed to be connected to TX from device 2 (and TX from device 1 to RX from device 2)

[image]

But in Corne Keyboard electronic design I can see that the connection between both Pro Micro goes from pin RX of one of them to the pin RX of the other one (green line). I read that pin RX can only receive data but can't send. So how do both Pro Micro communicate with each other?

[image]

Thanks in advance

getarun, 2021-07-01:
Hello, stumbling across this - I would like to confirm for the actual .sch file, but cant' open project's .sch. 
Tried Kicad 5.1.5. and some online tools.

foostan, 2021-07-10:
Split keyboard communication is implemented by QMK. There is no problem in actually connecting RX1s to each other.

thmxv, 2021-09-06:
I had to search a bit through th QMK doc to actualy find the technical explication, which can be resumed to:

> Serial in this context should be read as sending information one bit at a time, rather than implementing UART/USART/RS485/RS232 standards.

Basically QMK uses one GPIO pin to transmit data serially using a dedicated protocol, and have nothing to do with "classic" UARTs except the unfortunate 'serial' name.

The link given by @foostan is to the source code and quite difficult to understand. Those pages in the documentation are much more helpfull:
https://docs.qmk.fm/#/feature_split_keyboard
https://docs.qmk.fm/#/serial_driver

foostan, 2021-10-30:
I'll close it because it's not this project.

closed 2021-10-30 by foostan