foostan/crkbd #109: Right half fail with VIA firmware

ivanguardian, 2021-06-24:
Hi!!!
Could you help me to understand this case.

Right half, it does not work when I flash the pro micro with the firmware "VIA" , but it does works when I use one firmware generated from QMK

Left half
Pro Micro: red sparkfun
Firmeware: works with both (VIA,QMK)
Interface: USB C

Right half
Pro Micro: Pro Micro Black.
Firmware: works just with QMK.
Interface: micro USB.

firmware used:
QMK: crkbd_rev1_common_layout_split_3x6_3_mine.hex
VIA: crkbd_rev1_common_via.hex

[image]

keyboard-magpie, 2021-06-24:
What exactly doesn't work?

No function at all with the black micro?

ivanguardian, 2021-06-25:
> What exactly doesn't work?
> 
> No function at all with the black micro?

is like nor recognize the firme ware, and it

closed 2021-06-25 by ivanguardian

ivanguardian, 2021-06-25:
> What exactly doesn't work?
> 
> No function at all with the black micro?

That's right, absolutly nothing work
Even it causes a windows error "Usb device not recognized"

So that's why I had to get a red pro micro, in order to test instead of the left side black pro micro. 
It turned out without a hitch.
[image]