foostan/crkbd #110: Type of PCB

luifer10, 2021-06-24:
Hi i'm new in all of this and i want to make my own PCB in a fabric but they told me the thickness and the material sombody could help me with that?

And also i'm trying to understand whcih code is the correct one for v3 cherry

waymond91, 2021-06-27:
I think you're way better off posing this question to some electronics forums than here - the pro-micro used in the project is an atmega32u4 - so I would try on the arduino forum. Its super active.
Getting PCBs manufactured is also a good question for an electronics forum - not really a subject that can be covered comprehensively here. Maybe start with EEVBlog tutorials.

foostan, 2021-07-10:
@luifer10 Basically, you can use 1.6mm PCB. If you want to change thickness you can use 1.0mm - 1.6mm. Thanks.

closed 2021-07-10 by foostan