foostan/crkbd #111: Board outline is missing or malformed.

waymond91, 2021-06-26:
Fine work! I love the project and I'm super excited to build my own.
I am trying to generate gerber using KiCAD 5.99 however I cannot get the edge cuts to close. 
So that none of the 3d renderings by KiCad or manufacturers seem to come out correctly.
[image]
[image]

Any advice would be great :)

641i130, 2021-08-18:
They still haven't solved this huh.

foostan, 2021-08-24:
Thank you for reporting!
I fixed the classic, and cherry is no problem.
[image]

Rendering results from some PCB vendors may not display correctly. For example, the display of jlcpcb is different from the result of kicad.

closed 2021-08-24 by foostan

LoyalPotato, 2022-02-10:
Hey, I'm still getting the pcb like the first picture in the issue.
I'm using kicad v6.0.1. When I open the plot menu, I get a 
> Global solder mask minimum width and/or margin are not set to 0. Most board manufacturers expect 0 and use their own constraints for solder mask minimum width.

I'm not sure if it has anything to do with the issue (I don't really know what it means, because I'm new to this, just trying out).
I generate the drill files and then plot.
Then upload the .zip into jlcpcb and it still shows with filled areas, like so:

[image]