foostan/crkbd #114: more information

R4wm, 2021-07-12:
it would be helpful if we could include information for low production (hard to find) parts..
for example: where would someone source the low profile adhesive foam from? 
re: https://github.com/foostan/crkbd/blob/master/corne-light/doc/v2/buildguide_low_edition_en.md#parts

Eloren1, 2021-08-14:
Also why is it written in the requirements about the plate and the bottom part, if in this case they are not used? Or are they still needed?

Eloren1, 2021-08-14:
And can I use plates for the standard Corne or do I need another plate for the low edition?

Eloren1, 2021-08-14:
And can I use plates for the standard Corne or do I need different plate for the low edition?

keyboard-magpie, 2021-08-14:
@Eloren1 the only specific for the plates for a low profile choc build is that they should be 1.2-1.5mm thick.

cburg, 2021-08-24:
Based on some reddit comments and personal research, I'm leaning towards a soft silicone or polyurethane foam, and intuitively, I'm thinking open-cell foams will conform to the diodes / switch prongs.

In the photos, the foam appears to be about the same thickness as the PCB which is ~1.6mm or ~1/16". I measured the diodes on one of my boards and they sit about 1.15mm high. The leads from the pro-micro are ~1.3mm high.

I've got a few silicone foam samples on order and will post back if they seem to fit the bill, but it would be great if @foostan can quickly weigh in on this :).

foostan, 2021-08-24:
This form is this https://jp.misumi-ec.com/vona2/detail/110300286150/?HissuCode=PRGCS3-400-300
Sorry, I have information about Japanese store only.

Eloren1, 2021-08-24:
Ok, so it's 3mm thick rubber foam (easily found it on AliExpress)

cburg, 2021-08-24:
Thanks @foostan!

And for others, The MiSUMi appears to have other distribution centers globally as well. Here's a link to the same product on their US site:
https://us.misumi-ec.com/vona2/detail/110300286150/?HissuCode=PRGCS3-400-300&PNSearch=PRGCS3-400-300&searchFlow=results2type&KWSearch=PRGCS3-400-300

Just ordered some and it looks like it will get shipped from California.

foostan, 2021-08-25:
Thank you for your information!

closed 2021-08-25 by foostan