foostan/crkbd #123: Using mouse-key in VIA firmware

RyogaOTAKE, 2021-09-27:
I wanted to set mouse keys in my keymap using VIA. However, it was not worked. Is it impossible to use mouse keys through VIA app?

keyboard-magpie, 2021-09-27:
You will need to recompile the via firmware adding  `MOUSEKEY_ENABLE = yes` to the rules.mk file.

foostan, 2021-10-30:
I'll consider turning on this option for default.

jtr109, 2022-09-05:
> You will need to recompile the via firmware adding `MOUSEKEY_ENABLE = yes` to the rules.mk file.

To the newbie like me: This means you should download [the QMK firmware project](https://github.com/qmk/qmk_firmware) and change it in the [keyboards/crkbd/keymaps/via/rule.mk](https://github.com/qmk/qmk_firmware/blob/master/keyboards/crkbd/keymaps/via/rules.mk). Then, you should [flash it manually](https://github.com/foostan/crkbd/blob/main/doc/firmware_en.md#optional-if-you-build-the-firmware-yourself).

kamaaina, 2022-12-24:
rev3 of PCB seems to be out, but QMK only shows rev1 and r2g... how would one compile for rev3?

waffle87, 2022-12-31:
Yes, the `rev1` firmware in QMK is used for all versions of the DIY PCBs.

closed 2024-03-16 by foostan