foostan/crkbd #131: Hi,

foostan, 2021-10-30:
Hi,

I was wondering if it's possible to enable the Backlighting LEDs without soldering the Underglow LEDs, by directly connecting some pins. From what I understand the LEDs are chained, so that the Backlighting LEDs only work when the Underglow LEDs are also installed, since they are coming first in this chain. But by adding a connection between some pins of the first Underglow LED and the first Backlighting LED, it should be possible to power all the Backlighting LEDs, right?
I have a Corne Cherry v3.

Thank you in advance!

_Originally posted by @carlvr in https://github.com/foostan/crkbd/discussions/115_

agusdmb, 2022-10-12:
this is exactly what i did... i soldered the backlight without soldering the underglow leds... i bridged them and changed the firmware to have the correct support for RGB Matrix

i can think of two solutions here

1. Change the order of the leds, so that the first ones to solder are the backlight and then the underglow
2. Add some pads that can be shorted to skip the underglow leds (instead of having to manually bridge them)

foostan, 2024-03-16:
The undergrow led is deprecated
https://github.com/foostan/crkbd/releases

closed 2024-03-16 by foostan