foostan/crkbd #135: SPLIT_HAND_PIN for cherry v3

thmxv, 2021-11-11:
Hi,

I was wondering if the cherry v3 PCBs have a pin that is always low or high depending on the hand of the PCB? The default firmware binary does not work correctly if the right hand is connected to USB and setting this option in the firmware looks like the easiest and most flexible option to correct the issue.

keyboard-magpie, 2021-11-11:
They do not. The easiest thing to explore is either setting EE_hands or Master_right. Search "qmk docs handedness".

thmxv, 2021-11-11:
Thanks for the answer. I read the doc already and I am aware of all the options. I just did not knew if crkbd had hardware support for this feature. I suppose soldering a cable on the  underside of the PCBs is also an option, albeit not the cleanest looking.

foostan, 2021-11-23:
For users who use VIA etc. and do not edit the firmware directly, flag control like Master_right may be troublesome, so consider using SPLIT_HAND_PIN.

foostan, 2024-03-16:
v4 is released and `SPLIT_HAND_PIN` is enable
https://github.com/foostan/crkbd/releases

closed 2024-03-16 by foostan