foostan/crkbd #136: Flashing VIA firmware without using qmk_toolbox?

jberryman, 2021-11-12:
I'm really new to all this, and in the middle of a build. I'm on linux which doesn't support qmk_toolbox and following along these guides for my build:

https://github.com/foostan/crkbd/blob/main/corne-chocolate/doc/buildguide_en.md
https://github.com/foostan/crkbd/blob/main/doc/firmware_en.md

I know this is probably very basic, but it would be super helpful to have a not in the latter document about how to flash on linux. It seems like I'm maybe meant to use `dfu-programmer`, but I'm not sure and don't want to screw anything up.

jberryman, 2021-11-13:
It looks like this worked fine for the elite C, after pressing reset:

```
dfu-programmer atmega32u4 erase 
dfu-programmer atmega32u4 flash crkbd_rev1_via.hex 
dfu-programmer atmega32u4 reset
```

foostan, 2021-11-23:
The manual flash method has already been described. Please try to do with the QMK doc.
https://github.com/foostan/crkbd/blob/main/doc/firmware_en.md#optional-if-you-build-the-firmware-yourself

closed 2021-11-23 by foostan

TomzBench, 2022-02-09:
@foostan does this method work for the elite-c micro?

When flashing with `make crkbd/rev1/common:via:avrdude` I just get the dot's and never complete. The LCD seems stuck reading the default screen. Keys are ignored. and the reset button doesn't do anything...

TomzBench, 2022-02-09:
from @jberryman dfu-programmer worked. Perhaps add to firmware doc

foostan, 2022-02-09:
@TomzBench elite-c's bootloader is DFU, not avrdude.
https://keeb.io/products/elite-c-low-profile-version-usb-c-pro-micro-replacement-atmega32u4

mbledkowski, 2022-11-28:
> It looks like this worked fine for the elite C, after pressing reset:
> 
> ```
> dfu-programmer atmega32u4 erase 
> dfu-programmer atmega32u4 flash crkbd_rev1_via.hex 
> dfu-programmer atmega32u4 reset
> ```

I get the following error message:
```
> dfu-programmer atmega32u4 erase
dfu-programmer: no device present.
```
I have a Arduino ProMicro clone.