foostan/crkbd #139: SK6812MINI-E for easier soldering on Corne Choc?

jberryman, 2021-11-14:
(feel free to close if this isn't appropriate)

I heard about this part (https://www.youtube.com/watch?v=oLYkzn0jwg8) which IIUC is an addressable LED about the same size as the SK6812MINI, but with leads that would make it easier to solder (in particular for the per-key LEDs which I am really struggling with; by far the hardest thing on the Corne Choc build

Unfortunately for some reason the pinout doesn't align with the pads on the current PCB. Just thought it might be interesting to consider if you do another revision.

foostan, 2021-11-23:
Thank you for your comment.

In the future, Corne Choc will also use the SK6812 MINI-E.

maverick1872, 2022-10-24:
Any idea when we might get a V3 based Choc variant? Getting ready to build one myself and would love it to be based on the latest design. Subsequently it would make my day if it was adjusted to have Choc spacing instead of the MX spacing.

foostan, 2024-03-16:
https://github.com/foostan/crkbd/releases includes choc version.

closed 2024-03-16 by foostan