foostan/crkbd #145: staggered pinky like kyria with kailh choc

rnyPlanet, 2021-12-18:
staggered pinky like kyria. it can turn out to take as a [basis](https://github.com/foostan/crkbd/blob/main/corne-light/doc/v2/buildguide_low_edition_en.md)?

will it be possible to use kailh choc without hot swap for [this keyboard](https://github.com/raychengy/fifi_split_keeb#fifi-keyboard--%EF%B8%8F) which is based on the crkbd (I would try to do it myself - but I have no knowledge) ?
[image]

and can you explain what these holes are for?
[image]

@foostan maybe it won't take you long

keyboard-magpie, 2021-12-18:
In your first picture, no, that will require redesign for choc specific sockets.

In your second picture, the 3 holes you have circled are for MX, Alps, and the rightmost is the extra leg for choc v2 switches.

rnyPlanet, 2021-12-18:
> In your first picture, no, that will require redesign for choc specific sockets.

Is it possible, on the basis of the second board, to make a staggered for the pinky as in the first photo? (no sockets - to solder the switch to the board)

keyboard-magpie, 2021-12-18:
It wouldn't be a crkbd then...

Its open source for a reason :) modify away.

rnyPlanet, 2021-12-18:
> It wouldn't be a crkbd then...

I wore here because I like crkbd) but I would like a little improvement

> Its open source for a reason :) modify away.

I have no knowledge (

keyboard-magpie, 2021-12-18:
I would invite you to come and talk on the low profile keyboards server, we have a #design channel [https://discord.gg/69UM9sQF8q](url)

rnyPlanet, 2021-12-18:
> I would invite you to come and talk on the low profile keyboards server, we have a #design channel [https://discord.gg/69UM9sQF8q](url)

oh, interesting. I'll take a look)

foostan, 2021-12-26:
@rnPlanet Thank you for sharing your design, I hole you challenge yourself to design your own 👍 
@keyboard-magpie Thank you for everything 🙏

closed 2021-12-26 by foostan

rnyPlanet, 2021-12-26:
> I hole you challenge yourself to design your own 

thanks😁  could you look with one eye and tell me if it will work?) https://github.com/pashutk/chocofi/pull/2, @keyboard-magpie or maybe you have knowledge)