foostan/crkbd #146: SK6812MINI-E - No Caps/Resistors?

BeefcakeTheMighty, 2021-12-22:
I can't think of a way to phrase my question that doesn't sound critical so I'll preface this by saying that my question is **not** intended to be critical of your work. 

Why did you depart from the "typical application" illustrated on the data sheet by excluding the resistors and capacitors? 

Thanks in advance!

BenRoe, 2021-12-22:
I have build a lot keyboards with SK6812 Mini-E and it worked without the capacitor, or resistors.

BeefcakeTheMighty, 2021-12-23:
I'm not necessarily saying there should be a problem. I'm just looking for clarity about how this application differs from a "typical" application.

Or maybe the "typical" application recommends caps and resistors out of an abundance of caution??

keyboard-magpie, 2021-12-23:
It's a balance between functionality and ease of building and component count. It works without them, absolutely fine, for years- so why add them if just to appease the data sheet writers?

BeefcakeTheMighty, 2021-12-23:
Fair enough. Thanks for the guidance!!

foostan, 2021-12-26:
On keyboards, I often see cases where the SK6812MINI-E is used without Caps or Resistors. Theoretically, it might be better to have them, but we use them because they have a proven track record.

closed 2021-12-26 by foostan