foostan/crkbd #149: Improved Bluetooth Support

cburg, 2022-01-07:
While the Corne supports the Nice!Nano, there isn't support for an on-off switch or a battery connector, and battery support is quite limited. I really do not like the idea of a permanently soldered battery. 

I'm willing to attempt to make these changes myself, but I've never used KiCad before and my time is quite limited.

# LiPo Battery
- Recommendation: 301230 (3mm x 12mm x 30mm) --> 501235 (5mm x 12mm x 35mm) depends greatly on socketing / wire routing. Must be >= 100 mAh for the Nice!Nano. 

The Nice!Nano comes with a 301230 (3mm x 12mm x 30mm) 110 mAh battery. This fits under the Nice!Nano if you socket it *with* a taller pin row. For reference I used the mill-max 335 low profile interconnect to get an extra 0.7mm to account for the LiPo wires.
https://www.mill-max.com/products/socket/335-xx-xxx-00-160000/335-40-132-00-160000?s_term=335-40-132-00-160000&s_type=Quick%2FProduct%2FPart+Number+Search

A taller pin row (e.g. mill-max 800 interconnect) should accommodate a 4mm thick battery. And an even taller pin row (e.g. mill-max 350 interconnect) should accommodate a 5mm thick battery. 

This allows for quite a larger range so people can source batteries easier. For example, the Nice!Nano comes with 301230, but in the U.S. it's easier to source a 401230 from adafruit.

One more possibility, would be to have a cut-out under the microcontroller to support routing a battery under the PCB for thicker cases. I think this is going to be questionable since hotswap sockets take 2 mm of space, and for soldered switches, you need space / protection to prevent the exposed pins from damaging the lipo battery skin. Accommodating that extra space will mean ~9.1mm between the bottom of the PCB and a desk (2mm hot swap + 3mm battery + 1.6mm bottom plate + 2.5mm bumper). Any additional shielding will increase that thickness. But, this would allow batteries that are > 500 mAh and should offer months between charges. 

# Battery Connector
- Recommendation: Molex PicoBlade

For the battery connector, I think supporting the Molex PicoBlade system would be best. These are quite compact, and you can use one battery connector to connect to both board mount headers as well as cable to cable connectors. The board mount headers have both through-hole and surface mount and appear easy to hand solder. These connectors support 26-32 gauge wire with a single housing (two different crimp sizes though) and will cover the range of LiPo batteries I have come across that will fit on the Corne.
https://www.molex.com/molex/products/family/picoblade

# Power Switch
- Recommendation: 7 pin 3 terminal surface mount right angle slide switch. 

These seem pretty commonly used in the mechanical keyboard market (e.g. Nice60 and Ferris Sweep). They are quite thin at only 1.42mm high which makes them thinner than the hot-swap sockets as well as rubber bumpers / foam pads. They are also quite easy to source. 

# Placement of Parts
My main goal here is to keep the same form factor so existing cases will mostly fit. There are 2 layouts I'm thinking of. A Full Layout which retains full capabilities (e.g. one board for either bluetooth or wired), but has much tighter fits for components. A BLE only layout which removes unnecessary components (TRRS + OLED) for better spacing. 

- LiPo: 
  - **Full Layout**: Under the Nice!Nano. The board is 35 mm long if the LiPo goes under the USB connector as well or about 30 mm otherwise.

- Power Switch: 
  - **Full Layout**: Bottom of the board under the USB connector. These switches are only 1.4mm and are thinner than the hot-swap sockets. This is the positioning used by the Ferris Sweep. Looking at pictures, this positioning should be compatible with the boardsource LP aluminum case (I have one on order and will test), and it may be compatible with the IMK case from keyhive. I don't have a good way to verify though. For bare plates with rubber bumpers or the foam pad there will be enough clearance as well. 
  - **BLE Only Layout**: Layout which removes the TRRS jack and OLED headers. Power switch would then go next to the reset swtich.
  
- Battery Connector:
  - **Full Layout**: Move the OLED header to be flush with the TRRS jack, and place a PicoBlade right angle connector with the opening facing the microcontroller. This is going to be a very tight fit, but I think would work. This placement will only be compatible with 30mm long batteries, and the battery may also need to sit under the USB connector to avoid too much stress on the wires at the terminals. 
  - **BLE Only Layout**: Layout which removes the TRRS jack and OLED headers. PicoBlade right angle connector will occupy the space where the TRRS and OLED headers were.

foostan, 2022-04-07:
@cburg I do not have Nice!Nano so I cannot verify this. Are there any other examples of implementations or published data that would be helpful?

cburg, 2022-06-20:
@foostan, there are a few references for wiring 

* **Ferris Sweep** (switch under Nice!Nano)

    https://github.com/davidphilipbarr/Sweep/tree/main/Sweepv2.2

* **Keyboard Kitchen** (hacked switch onto TRRS pads, looks like what @ykz89  did on their build)

    https://keyboard.kitchen/2021/03/28/build-log-corne-chocolate-v2-1/

* **Nice!Nano Schematics**

    https://nicekeyboards.com/docs/nice-nano/pinout-schematic/

I'm partial to the Sweep's implementation since it will still allow a single USB design.

HubbeKing, 2022-06-20:
To be honest, if you put a power switch footprint on the board, there's not a ton of reasons to also add a battery jack. Routing the power switch to a pair of througholes for B- and B+, like the Ferris Sweep does, is perfectly fine. Replacing the battery should be a fairly uncommon occurrence, and desoldering a pair of wires to do that isn't too bad IMHO.

sungo, 2022-06-20:
@cburg I've been playing with some options and, regardless of height concerns, I think the bigger problem is the connectors themselves. JST-PH, picoblade, all of them are really tight fits for the battery cable itself. I worry about folks tearing off the connector while trying to unplug the battery. I've pulled the wires out of a jst or picoblade a few times. (Admittedly my grip slipped.)

To @HubbeKing 's point, if B-/B+ are exposed via through holes, folks can connect however they want. I can attach a jst-ph, you can attach a picoblade, other folks can direct connect.

it should be pretty easy to fit a pair of through holes in the v3. More challenging in the v2 based classic for choc folks but my understanding is that the v4 choc board will be based on the v3.

cburg, 2022-12-23:
So, I recently discovered this:

https://github.com/petejohanson/crkbd/tree/board/corne-ultralight/corne-ultralight/pcb

Looks like the creator of the ZMK firmware went and modded the corne-light board to not only support a JST battery connector and power switch, but also the new Nice!View display header which requires 5 pins instead of the standard 4:

https://nicekeyboards.com/nice-view/

I would still prefer a single board that could support both bluetooth and wired, but there's simply not enough space for both the JST header and TRRS connector. 

At the end of the day, this is an excellent alternative.

closed 2023-11-11 by foostan