foostan/crkbd #151: Corne Mini SVG Wrong Size?

articnova, 2022-01-22:
Is the corne mini SVG file exported with the wrong dimensions?  Hole size for choc is around 10mm instead of the 13.851mm that the corne mini tent SVG has.

keyboard-magpie, 2022-01-22:
SVG scaling is set in the user’s environment so check your own settings and calibration. (Mm vs inch etc).

articnova, 2022-01-22:
Sorry yeah seems like there is a fusion 360 issue with importing SVGs. Lots of recommendations to import using 1.33 scale.

https://www.reddit.com/r/Fusion360/comments/ldkdns/importing_svg_svg_is_smaller_than_original/
https://www.reddit.com/r/Fusion360/comments/iggn0d/scaling_problem_importing_svg_from_illustrator/
https://www.reddit.com/r/Fusion360/comments/eqy8y1/weird_annoyance_when_importing_svg_files/

closed 2022-01-22 by articnova