foostan/crkbd #166: Number the LEDs on cherry silks in wiring order

grzes, 2022-04-10:
Seems like the first backlight LED after the last underglow is the outermost thumb cluster one and from there the strip "snakes" through the columns, but the silk print numbers the leds in rows instead. It would mahe troubleshooting easier if the labels matched the strip wiring order.

agusdmb, 2022-10-12:
i soldered and soldered the LED 1 until i completely distroyed it waiting for it to turn on... i then realized the numbers were wrong :facepalm:

artt, 2022-12-12:
That threw me off too until I found this: https://github.com/foostan/crkbd/issues/96

Maybe I'll add this to the build guide.

foostan, 2024-03-16:
https://github.com/foostan/crkbd/releases uses matrix led and v4 is recommended PCBA so it doesn't care silk order.

closed 2024-03-16 by foostan