foostan/crkbd #167: About acrylic sheet thickness of corne-cherry

LuckyFishGeek, 2022-04-15:
PCB: corne-cherry
plates: common

What is the thickness of acrylic ? Sorry I'm a newbie asking very elementary questions!
Thanks for the help

kjunuh, 2022-04-15:
Generally non-acrylic plates are 1.5mm as that's what switches are normally designed to snap into, but for acrylic people normally recommend going with 3mm because 1.5mm acrylic cracks very easily. The trade off is that the switches won't snap into the plate if you go with 3mm. Either is viable but you do have to be extra careful with 1.5mm plates.

LuckyFishGeek, 2022-04-15:
> Generally non-acrylic plates are 1.5mm as that's what switches are normally designed to snap into, but for acrylic people normally recommend going with 3mm because 1.5mm acrylic cracks very easily. [...]

Thank you for your kind help and wish you a happy life

closed 2022-04-15 by LuckyFishGeek