foostan/crkbd #170: fireware  issues

LuckyFishGeek, 2022-04-21:
PCB: corne-cherry V3

After the crkbd_rev1_common_via.hex firmware is flashed into both the left and right ProMicro devices, only one side works at a time, the other keyboard does not work and the OLED does not display.

Specific demonstration

 https://youtu.be/5wZfeNqG4VE

keyboard-magpie, 2022-04-21:
Pro Micro controllers have a hardware bug that causes this issue. 

You will need a QMK command line environment set up and working. 

You need to edit your keymap's config.h file to add `#define SPLIT_USB_DETECT` and recompile and reflash the firmware. 

Alternatively, you can flash this firmware which already has these changes made: https://qmk.tzarc.io/crkbd_r2g_via.hex

closed 2022-04-21 by LuckyFishGeek

psclwsr, 2023-01-06:
I tried using this firmware file, but the provlem still persists. I am using the ELITE-C controllers. The sides work on their own, only the one connected via TRRS doesnt.

lexa2hk, 2023-03-21:
thanks! helped a lot

TRIPSLORD, 2023-06-27:
> Pro Micro controllers have a hardware bug that causes this issue.
>
> You will need a QMK command line environment set up and working.
>
> You need to edit your keymap's config.h file to add `#define [...]

The link is dead can anyone reupload?

BlwAverage, 2023-07-17:
> The link is dead can anyone reupload?

If anyone else stumbles on this and needs it you can get my firmware file here https://github.com/BlwAverage/corne-kb-firmware