foostan/crkbd #171: Found a very strange LED problem

LuckyFishGeek, 2022-04-27:
PCB:corne-cherry V3   
Controller:ProMicro 

The keyboard worked for a while and then the right part of the LED stopped working, when I cleaned it with alcohol it worked again

I've checked the LED soldering and other components and they're fine   

Requesting your help.

 demo: https://youtu.be/r_4ABb2CrF0

foostan, 2022-04-28:
I think it's an LED failure or an assembly mistake.  Is it possible to investigate from where to where the LEDs work properly?  Also, please ask about this problem at Discord's support channel for troubleshooting.

https://discord.gg/eKC2ASsk

Thanks.

LuckyFishGeek, 2022-04-28:
> 

> I think it's an LED failure or an assembly mistake. Is it possible to investigate from where to where the LEDs work properly? Also, please ask about this problem at Discord's support channel for troubleshooting.
> 
> https://discord.gg/eKC2ASsk
> 
> Thanks.

solved thanks for the help

closed 2022-04-28 by LuckyFishGeek