foostan/crkbd #172: 3x5

illmillrig, 2022-05-13:
can this pcb not have the ends snapped off to be 3x5 anymore?  any physical reason why you couldn't cut the pcb?

waffle87, 2022-05-22:
Left half shouldn't be an issue if you were to cut down the middle.
[image]
Right half, there's a data line for the underglow LEDs.
[image]
This wouldn't be an issue if you don't plan to use RGB, or you could just replace the connection with a jumper wire.

illmillrig, 2022-05-22:
thank you.  I just scored it with a razor blade and snapped it like glass.  jumper wire worked perfectly.

closed 2022-05-22 by illmillrig