foostan/crkbd #173: Handedness, setting right side as master

raveltan, 2022-05-16:
Hello,

I am currently having a device with right's usb port starting to get loose, and want to try to move the usb connection to the right side of the device,

I am aware we can set this on qmk [`](https://docs.qmk.fm/#/feature_split_keyboard?id=setting-handedness), but upon trying, i'm unable to get it working, is there anyone who has gotten any success with this and are willing to share the configuration, 

Thank you

AlanMauricioC, 2022-05-16:
Hi there! here I'm using the right half as a master! 
on  your `config.h` file you should define this variable
`#define MASTER_RIGHT`
[here](https://github.com/AlanMauricioC/qmk_firmware/blob/master/keyboards/crkbd/keymaps/tangananica/config.h) you can check my personal config, if it helps

AlanMauricioC, 2022-05-16:
also, don't forget that you should compile the configuration and then flash both sides of the keyboard

raveltan, 2022-05-16:
@AlanMauricioC Thank you!

But upon trying, my right keyboard has become the master but the left one is now not sending any keys, 
if I try connecting the left one, they both become inverted but both are sneding keys,

do you know what am I doing wrong?

AlanMauricioC, 2022-05-16:
are you able to swap both micros? maybe one didn't get flashed correctly

raveltan, 2022-05-16:
Hmm, unfortunately it's soldered and I'm don't have an access to a solder at the moment, do you have any other ideas of what I can try at the moment?

AlanMauricioC, 2022-05-16:
that's supper weird, :/ bc when you connect the left one, it works as expected, sorry but I think I can't help any more

foostan, 2022-06-07:
I don't think this is a design issue, so could you continue to discuss on my discord server.

closed 2022-06-07 by foostan